ovh/overthebox-feeds #9: VMWARE esxi 5.5 boot error

merydwin, 2016-03-22:
Hi i used the .ova for openwrt and i used the command

` sudo dd if=openwrt-x86-64-combined-ext4.img of=/dev/mmcxxx`

and at boot i have an message 
http://puu.sh/nQl3k/952a0eee85.png

i tried almost everything but no progress :/

merydwin, 2016-03-23:
Update: I tested my vmdk on virtual box and it's working fine i think it's an hardware incompatibility
Update 2 : Base openwrt work on vmware esxi

simlelievre, 2016-04-05:
Hello Merydwin,
I'm adding modules in kernel. Could you send me list of module used in your base openwrt on vmware esxi ?

```
root@OpenWrt:~# lsmod
```
Thanks for your help,
Simon

merydwin, 2016-04-08:
Thks i will do it in few hours

simlelievre, 2016-04-08:
and also 
```
zcat /proc/config.gz
```

merydwin, 2016-04-08:
Ls mod 
`crc_ccitt               1088  1 ppp_async
e1000                  89216  0
e1000e                135056  0
ip6_tables              8536  3 ip6table_raw
ip6t_REJECT             1280  2
ip6table_filter          800  1
ip6table_mangle          960  1
ip6table_raw             704  1
ip_tables               8680  4 iptable_nat
ipt_MASQUERADE           704  1
ipt_REJECT              1120  2
iptable_filter           832  1
iptable_mangle           928  1
iptable_nat              992  1
iptable_raw              736  1
ipv6                  265633 28 nf_conntrack_ipv6
mii                     2976  1 r8169
nf_conntrack           47336 11 nf_nat_ipv4
nf_conntrack_ftp        6000  1 nf_nat_ftp
nf_conntrack_ipv4       5376  8
nf_conntrack_ipv6       5696  3
nf_conntrack_rtcache    2304  0
nf_defrag_ipv4           928  1 nf_conntrack_ipv4
nf_defrag_ipv6         21320  1 nf_conntrack_ipv6
nf_log_common           2272  2 nf_log_ipv4
nf_log_ipv4             3424  0
nf_log_ipv6             3904  0
nf_nat                  8966  5 nf_nat_ipv4
nf_nat_ftp              1312  0
nf_nat_ipv4             3586  1 iptable_nat
nf_nat_masquerade_ipv4    1348  1 ipt_MASQUERADE
nf_reject_ipv4          1728  1 ipt_REJECT
nf_reject_ipv6          1888  1 ip6t_REJECT
ppp_async               5736  0
ppp_generic            20152  3 pppoe
pppoe                   7680  0
pppox                   1152  1 pppoe
pps_core                5364  1 ptp
ptp                     8348  1 e1000e
r8169                  56836  0
slhc                    4128  1 ppp_generic
x_tables                9832 26 ipt_REJECT
xt_CT                   2400  0
xt_LOG                   864  0
xt_REDIRECT             1152  0
xt_TCPMSS               2624  2
xt_comment               608 62
xt_conntrack            2368 10
xt_id                    608  0
xt_limit                1122 20
xt_mac                   704  0
xt_mark                  800  0
xt_multiport            1248  0
xt_nat                  1280  0
xt_state                 832  0
xt_tcpudp               1856 10
xt_time                 1696  0
`

merydwin, 2016-04-08:
lsmod
`
crc_ccitt               1088  1 ppp_async
e1000                  89216  0
e1000e                135056  0
ip6_tables              8536  3 ip6table_raw
ip6t_REJECT             1280  2
ip6table_filter          800  1
ip6table_mangle          960  1
ip6table_raw             704  1
ip_tables               8680  4 iptable_nat
ipt_MASQUERADE           704  1
ipt_REJECT              1120  2
iptable_filter           832  1
iptable_mangle           928  1
iptable_nat              992  1
iptable_raw              736  1
ipv6                  265633 28 nf_conntrack_ipv6
mii                     2976  1 r8169
nf_conntrack           47336 11 nf_nat_ipv4
nf_conntrack_ftp        6000  1 nf_nat_ftp
nf_conntrack_ipv4       5376  8
nf_conntrack_ipv6       5696  3
nf_conntrack_rtcache    2304  0
nf_defrag_ipv4           928  1 nf_conntrack_ipv4
nf_defrag_ipv6         21320  1 nf_conntrack_ipv6
nf_log_common           2272  2 nf_log_ipv4
nf_log_ipv4             3424  0
nf_log_ipv6             3904  0
nf_nat                  8966  5 nf_nat_ipv4
nf_nat_ftp              1312  0
nf_nat_ipv4             3586  1 iptable_nat
nf_nat_masquerade_ipv4    1348  1 ipt_MASQUERADE
nf_reject_ipv4          1728  1 ipt_REJECT
nf_reject_ipv6          1888  1 ip6t_REJECT
ppp_async               5736  0
ppp_generic            20152  3 pppoe
pppoe                   7680  0
pppox                   1152  1 pppoe
pps_core                5364  1 ptp
ptp                     8348  1 e1000e
r8169                  56836  0
slhc                    4128  1 ppp_generic
x_tables                9832 26 ipt_REJECT
xt_CT                   2400  0
xt_LOG                   864  0
xt_REDIRECT             1152  0
xt_TCPMSS               2624  2
xt_comment               608 62
xt_conntrack            2368 10
xt_id                    608  0
xt_limit                1122 20
xt_mac                   704  0
xt_mark                  800  0
xt_multiport            1248  0
xt_nat                  1280  0
xt_state                 832  0
xt_tcpudp               1856 10
xt_time                 1696  0
`

merydwin, 2016-04-08:
And /proc/config.gz doesn't exist

simlelievre, 2016-04-19:
Hello,
We recompile a new kernel with lot of option for controllers.
Could you test image on http://downloads.overthebox.ovh/testing/x86/64/ ?
Simon

merydwin, 2016-04-19:
Ok so working fine :) but no ipv4 only ipv6 and https://overthebox.ovh/ give me ERR_CONNECTION_REFUSED

merydwin, 2016-04-24:
Ok so everything is working, i convert the VDI file to VMDK with virtualbox but once i go on the web admin i see 
http://puu.sh/ouxur/b421f0dc8e.png

Even if i change the password everything is reseted

merydwin, 2016-04-24:
Edit : if i try this way (https://docs.ovh.com/pages/releaseview.action?pageId=18121070) with the new img i still got the error about Waiting for root device PARTUUID=c0b6e087-02...

merydwin, 2016-04-25:
So all other img is not working.
Converting the VDI to VMDK with virtualbox work but it's readonly mode
Converting the VDI to VMDK with "qemu-convert" work perfectly :)

sduponch, 2016-04-25:
Can you check with mount if you have a partition marked as read only ?
---
function checkReadOnly()
         for _, mount in pairs(get_mounts()) do
                 if mount.mount_point and mount.mount_point == "/" then
                         if mount.options and mount.options:match("^ro") 
then -- assume ro is always the first of the options
                                 return true
                         end
                 end
         end
         return false
end
---

closed 2016-06-14 by sduponch