ovh/overthebox-feeds #11: problème de WANIP

jlabrous, 2016-04-21:
De temps en temps l'adresse le l'OTB prend celle de mon ADSL.

J'ai fini par trouver une piste dans les iptables.

Quand il y a le problème la règle 'mwan3_hook' n'est plus la prèmière.
Ici j'ai 3 fichiers:
- iptables.good c'est un iptables-save juste après un reboot.
- iptables.bad c'est un iptables-save juste après l'apparition du bug
- iptables.cor c'est l'iptables.bad corrigé et restoré (iptable-restore)

Je teste avec "wget -O - -q http://ifconfig.ovh" 

Le problème arrive presque à chaque fois si je place les interfaces dans l'ordre if2,eth1,tun0,xtun0 dans la conf mwan3.
Ceci pour que les interfaces réels montent avant les tunnels.
Alors un simple 'restart mwan3' provoque le PB.  

La règle 'mwan3_hook' est ajoutés par /etc/hotplug.d/iface/15-mwan3
Les règles 'QOS_MARK' sont ajoutées par /usr/lib/sqm/otb.qos
A noté que l'auteur de simple.qos a déjà commenté la ligne.

Si ça peut vous aider...

Jean Labrousse

iptables.good:-A OUTPUT -j mwan3_hook
iptables.good:-A OUTPUT -o if2 -g QOS_MARK_if2
iptables.good:-A OUTPUT -o eth1 -g QOS_MARK_eth1
iptables.good:-A OUTPUT -j delegate_output

iptables.bad:-A OUTPUT -o if2 -g QOS_MARK_if2
iptables.bad:-A OUTPUT -j mwan3_hook
iptables.bad:-A OUTPUT -o eth1 -g QOS_MARK_eth1
iptables.bad:-A OUTPUT -j delegate_output

iptables.cor:-A OUTPUT -j mwan3_hook
iptables.cor:-A OUTPUT -o if2 -g QOS_MARK_if2
iptables.cor:-A OUTPUT -o eth1 -g QOS_MARK_eth1
iptables.cor:-A OUTPUT -j delegate_output

sduponch, 2016-04-21:
Merci, je regarde pour m'assurer d'avoir le mwan3_hook en premier en 
ajoutant les marques de QoS

jlabrous, 2016-04-21:
J'ai tenté une modif sur /etc/hotplug.d/iface/15-mwan3:
#               $IPT -A OUTPUT -j mwan3_hook
                $IPT -I OUTPUT 1 -j mwan3_hook
ça semble faire l'affaire ...
Jean

sduponch, 2016-04-22:
Mon correctif dans otb.qos

+    ipt -t mangle -A OUTPUT -o ${IFACE} -g QOS_MARK_${IFACE}
+    ipt -t mangle -I OUTPUT 2 -o ${IFACE} -g QOS_MARK_${IFACE}

sduponch, 2016-04-22:
Errata :

     # Marking dscp SS tagged packets
     ipt -t mangle -D OUTPUT -o ${IFACE} -g QOS_MARK_${IFACE}
     ipt -t mangle -I OUTPUT 2 -o ${IFACE} -g QOS_MARK_${IFACE}

closed 2016-05-23 by sduponch