ovh/overthebox-feeds #55: Boot Problem ESXI 6.0

loicll, 2016-06-26:
Hello,

I tried to convert the vdi image for OverTheBox into vmdk to use it with ESXI 6.0 but it never start, i always get a kernel panic.I tired modifying the type of lan card with the three esxi offer, still the same problem. I'm just starting with ESXI and OverTheBox so i might be doing something wrong
here is a screen capture of the error i get[image]
i used the "openwrt-x86-64-combined-ext4.vdi" image.
I was wondering if the problem isn't ESXI 6.0 cause apparently someone were able to make it work on ESXI 5.5
Thanks in advance for ur help

sduponch, 2016-06-29:
Hi,

can you try with this autogenerated VMDK ?
http://downloads.overthebox.ovh/develop/x86/64/openwrt-x86-64-combined-ext4.vmdk

Best regards,
DUPONCHEEL Sébastien

loicll, 2016-06-29:
Hello,

Thanks for that, but  i made it work one hour ago, i converted the openwrt-x86-64-combined-ext4.img into vmdk, i was still getting the kernel panic and then i converted the vmdk into another vmdk with a tool included in esxi, don't remember the name of it, finally got it to start and finally just had to execute the script wrote in the "read only" error msg, now it seem to work perfectly.waiting for my second xdsl connection to be created to test it and subscribe to overthebox

Regards,

closed 2016-06-29 by loicll

loicll, 2016-07-07:
Hello,
I actually still have problems with overthebox and esxi 6, the thing boot,  but nothing seem to behave like it should : no dhcp lease when i first start it, if i force it to get one (by setting it myself) it works put never detect the others connections plugged on the same switch, the dhcp server of the overthebox barely works, it's like none of the network configurations are working.Can someone try to make a vmdk (or give me one if it already exist) that work with esxi 6 ? cause i don't understand how everything is configured to do that myself 

(ps:i tried the auto generated VMDK given to me before, i get kernel panics)

Thanks in advance,