ovh/overthebox-feeds #71: Overthebox does not get dhcp lease on ESXi 5.5

rom85m, 2016-07-05:
Hello,
We want to install overthebox on a Vmware Infrastructure with ESXi 5.5.
We use the last image http://downloads.overthebox.ovh/trunk/x86/64/openwrt-x86-64-combined-ext4.img.gz
The VM boot without problem but it does not get DHCP lease. We have a Windows VM on the same ESXi, in the same network et this Windows VM get DHCP lease without problem.

We try another method : installation of overthebox on Vmware Workstation with the netword card bridged on a network with the box, overthebox gets DHCP lease and working. We shutdown the VM, we take files (vmx, vmdk ...) and we use Vmware Converter to convert the VM Workstation to VM of own Vmware Infrastructure ESXi 5.5. The converted Overthebox VM boot without problem, but it does not get DHCP lease.

Do you know a issue of this problem of getting DHCP lease ? Do you know a manuel method to initialize the overthebox VM on a Vmware Infrastructure with ESXi 5.5 ?

Thank you for your feedback.

loicll, 2016-07-07:
hello,
i had the same problem with esxi 6, i got it to work by using the command "uci set network.lan.proto=dhcp" and "/etc/init.d/network restart" it gets a ip address but the whole overthebox thing doesn't works, maybe my method is wrong tho, let me know if it worked for u or not 

Regards,

chrobiche, 2016-07-15:
I has the same issue as you. I finally got it to work.
The thing is that by default promiscuous mode is disable in Esxi. So with the vsphere client, connect to your host. Go to config > network > select the vswitch > properties > Vm network > security > Promiscuous mode > Accept.
And now everything should work.

closed 2017-06-16 by angt