ovh/overthebox-feeds #80: ESXI 6.0 Kernel panic

xeniluom, 2016-07-17:
Salut a tous,
j'ai créé une machine virtuelle sous ESXI 6.0 avec l'image : `http://downloads.overthebox.ovh/develop/x86/64/openwrt-x86-64-combined-ext4.vmdk` et l'image `http://downloads.overthebox.ovh/trunk/x86/64/openwrt-x86-64-combined-ext4.vmdk`
mais rien n'y fait, ca ne veut pas aller plus loin.
Niveau config, j'ai mis 2vCore, 2go de RAM pour la VM. Je travail sur un Z6000 ProLiant SL160z G6 de chez HP avec 2x Xéon E5506 et 36go de RAM. Un HDD de 250go et un second de 1TO. Le serveur passe par 2 switch 1000Mbps avant d'arriver à la Freebox. J'pense avoir fait la tour de la config.
[image]

J'ai également éssayé d'installer via un live CD d'ubuntu 16 et en téléchargeant directement l'image sur le disque, mais la il ne va pas plus loin que : `waiting for root device`
Auriez vous une piste? Mon matériel qui n'est pas compatible?
Merci.

chrobiche, 2016-07-18:
Change le type de disque pour IDE (pas SCSI).

xeniluom, 2016-07-18:
Merci pour le tuyaux, j'ai encore pas mal a apprendre sur linux/esxi pour mettre en oeuvre. Du coup je suis passé sur Workstation et effectivement, ça marche bien :) Par contre, j'crois que je vais galérer pas mal pour mettre au point le lien entre mon téléphone (en 4G) et la VM... (bientot un huawei b593 à la rescousse :) )

segobi, 2016-07-31:
Hi!
Login to your esxi host and convert the vmdk to zeroedthick:

vmkfstools -i openwrt.vmdk out.vmdk -d zeroedthick -a buslogic

closed 2017-06-16 by angt