ovh/overthebox-feeds #102: low speeds vs stock ubuntu mptcp

segobi, 2016-08-02:
Im only getting 100 kB/s per link (3 DSL Lines) opposed to 1800 kB/s per link on ubuntu using the same mptcp setup...

sduponch, 2016-08-02:
Hi,

Can you explicit your setup please (Hardware, OverTheBox service id...) ?

Best regards,
DUPONCHEEL Sébastien

segobi, 2016-08-02:
local mptcp setup without service - id - just testing downloads from 

http://multipath-tcp.org/snapshots/

its using all 3 links but speeds are really bad.
Same on self compiled openwrt with mptcp support in linux kernel.

I've been running this setup on bare-metal and on esxi and in virtualbox.

Setting up the same routes and multipath enabled devices its working without any problems on ubuntu. Seems to be some openwrt problem.

sduponch, 2016-08-02:
OverTheBox use virtual macvlan interfaces and this can cause issues with the ESXi network setup.

segobi, 2016-08-03:
hmm same problem running it on bare-metal. Are those if devices supposed to be macvlans of eth0 and then mptcp enabled + put in the mptcp firewall rule ? I put the appropriate gateways in the If devices config and overthebox seems to configure the correct routing... All three links are showing traffic on iptraf but just 100 kB/s.

closed 2016-08-04 by segobi

segobi, 2016-08-04:
No ESXi Problems - traffic over socks proxy working perfectly with mptcp... I should have done more tests :+1: