ovh/overthebox-feeds #128: Ping from Internet can break interface tracked by ping method.

jlabrous, 2016-08-30:
In track.lua, the case where a request is received instead of the ping response is not correctly processed.
"other error : 8". Type 8 is a "echo-request" not a response.
If a burst of ping is sent from internet to OTB public IP, Tun0 can be seen KO

Jean Labrousse

sduponch, 2016-08-30:
Link with #124

sduponch, 2016-08-31:
Fixed in 0.5.0 and later

simlelievre, 2016-09-05:
we are writing an envolved patch to fix it.

sduponch, 2016-09-05:
Link #136

jlabrous, 2016-09-05:
don't forget to manage EAGAIN(11) error from 'recv_from' too !

sduponch, 2016-09-05:
EAGAIN(11) is handled because we no longer "return" at first packet received (as we use RAW socket) but wait for the given timeout to expire.

sduponch, 2016-09-05:
Test package can be found here :
http://demold.ovh.net/download/460801b35e597d7d884be94e6bb2ed3c/mwan3otb_1.7-26_all.ipk

jlabrous, 2016-09-05:
> Test package can be found here :
> http://demold.ovh.net/download/460801b35e597d7d884be94e6bb2ed
> 3c/mwan3otb_1.7-26_all.ipk

Here my test result:

luabitop was missing (I had to build it )
After mwan3 restart
CPU was 92% sys and no trace about if2,tun0 or xtun0 pings

 *Mem: 263796K used, 1679856K free, 1496K shrd, 2868K buff, 28448K cached*

*CPU:   7% usr  92% sys   0% nic   0% idle   0% io   0% irq   0% sirq*

*Load average: 2.39 1.19 0.76 7/96 24714*

*  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND*

*21787     1 root     R     146m   8%  24% lua /usr/sbin/mwan3track -i if2
-d if*

*22339     1 root     R    96556   5%  24% lua /usr/sbin/mwan3track -i tun0
-d t*

*23950     1 root     R    61848   3%  24% lua /usr/sbin/mwan3track -i
xtun0 -d*

*21575 11169 root     S    11852   1%   1% {mwan3} /bin/sh /usr/sbin/mwan3
resta*

* 6055     1 root     S     133m   7%   0% /usr/bin/scollector -conf
/etc/scolle*

* 6125  6104 root     S    14656   1%   0% glorytun host 109.190.128.109
port 50*

I found ping was called from track.lua without size parameter. I put 4 just
like in your private branch
(The test package seems to not match this branch)

Now, it seems to work....I'll see tomorrow

I'm not sure it is a good idea to not test EAGAIN in receive_ping if data
is nil.
For any other reason the script can get 100%CPU until timeout stops the
crasy loop.
( the system call can return immediately if error is not EAGAIN increasing
the priority of the process )

Jean Labrousse

sduponch, 2016-09-06:
The test package was broken, i already have fixed the missing parameter.
i check for CPU usage

jlabrous, 2016-09-06:
The CPU issue was due to the missing parameter (urandom read with 'nil'
lenght )
Jean

sduponch, 2016-09-06:
fixed behaviour of missing size parameter

sduponch, 2016-09-06:
Prodded on 0.4.9

closed 2016-09-06 by sduponch