ovh/overthebox-feeds #142: newest trunk has error

debiansid, 2016-09-12:
```
Status: 500 Internal Server Error /usr/lib/lua/luci/dispatcher.lua:452: Failed to execute template dispatcher target for entry '/admin/overthebox/overview'. The called action terminated with an exception: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'overthebox/index'. A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'header'. A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'themes/ovh/header'. A runtime error occured: /usr/lib/lua/luci/ccache.lua:70: error loading module 'overthebox' from file '/usr/lib/lua/overthebox.lua': /usr/lib/lua/overthebox.lua:1049: unexpected symbol near '!' stack traceback: [C]: in function 'assert' /usr/lib/lua/luci/dispatcher.lua:452: in function 'dispatch' /usr/lib/lua/luci/dispatcher.lua:141: in function 
```

MartinWetterwald, 2016-09-12:
Thank you for the bug report
謝謝提交錯誤報告 :)

simlelievre, 2016-09-12:
fixed in #143

closed 2016-09-12 by simlelievre