ovh/overthebox-feeds #148: erreur à la compilation

clembert, 2016-09-14:
Bonjour,

Après avoir suivi les indications du paragraphe "Compile from source" pour installer le service overthebox sur un hôte ubuntu x86, et après avoir préparer les requis et lancer la compilation, j'obtiens ces messages d'erreur : 

./overthebox-openwrt/include/toplevel.mk:65: recipe for target 'prepare-tmpinfo' failed
make[1]: *** [prepare-tmpinfo] Error 25
./overthebox-openwrt/include/toplevel.mk:177 : la recette pour la cible « world » a échouée
make: *** [world] Erreur 2

Pouvez-vous m'aider ?

MartinWetterwald, 2016-09-14:
Bonjour,

Il manque des informations. L'erreur proprement dite n'apparaît pas dans le copier/coller.
Pouvez-vous copier/coller les erreurs après avoir lancé la compilation avec un seul thread et en mode verbose ? (-j 1 V=s)

clembert, 2016-09-14:
Bonjour,

Voici le résultats : 

Duplicate feed name 'overthebox', line: 7
Duplicate feed name 'overthebox', line: 7
Duplicate feed name 'overthebox', line: 7
./overthebox-openwrt/include/toplevel.mk:65: recipe for target 'prepare-tmpinfo' failed
make[1]: *** [prepare-tmpinfo] Error 25
./overthebox-openwrt/include/toplevel.mk:177 : la recette pour la cible « world » a échouée
make: *** [world] Erreur 2

debiansid, 2016-09-15:
check your feeds.conf, youhave duplicate feed name, just keep one.

closed 2016-09-27 by angt