ovh/overthebox-feeds #200: Wrong LAN traffic value

Rudloff, 2016-10-28:
Hello,

It seems if the request to `/cgi-bin/luci/admin/overthebox/lan_traffic_data` takes too much time, the displayed bandwidth usage will be too high.
I think this is because the value is divided by `period` [here](https://github.com/ovh/overthebox-feeds/blob/master/luci-app-overthebox/luasrc/view/overthebox/lan_traffic.htm#L138) even though it might have taken more time.

Wouldn't it be possible to calculate the kb/s server-side and return the computed value?

Regards,

Rudloff, 2016-10-28:
(Or we could have a timestamp in the JSON that would be used to calculate the interval.)

gregdel, 2017-06-20:
Hey @Rudloff, we have a lot of things on our plates... Sorry for the late reply.
To be honnest the lan traffic program is not in our priorities for now, but you can always make a PR on the project ! I'll be happy to take care of it :)

closed 2019-03-05 by angt