ovh/overthebox-feeds #220: What is function of this domain

debiansid, 2016-11-30:
provisionning.overthebox.net

MartinWetterwald, 2016-11-30:
This is a daemon that tells the OverTheBox what to do. The OverTheBox contact it periodically and asks what it should do.

Unfortunately we can't give you more details about the internals on OVH side.

closed 2016-11-30 by MartinWetterwald

debiansid, 2016-11-30:
I am wondering if you can provide the service to other cable company with your architecture services.

debiansid, 2016-11-30:
Such as I run tun and socks services on my own VPS and with my FTTB connection.

gregdel, 2016-11-30:
What do you mean ? You don't need this domain if you run your own infrastructure :)

debiansid, 2016-11-30:
Currently my local otb continues to connect this domain, and I can not run glorytun-Udp on it. Real time graph shows nothing. My idea is running  shadowsocks-libev for tcp and glorytun-udp for udp traffic. Would you please help me on that? Thank you Sir