ovh/overthebox-feeds #259: Support restricted networks

cbenhagen, 2017-01-10:
Currently OTB can not be used in networks that block foreign DNS servers and outgoing traffic on unusual ports like 4443 and 5006 and maybe others.

gregdel, 2017-06-20:
Sorry for the late reply, do you still have your issues ?
There's a new feature in our shadowsocks proxy (which only handles the TCP traffic of the OTB) which makes the DNS queries on our OVH servers and does not check the local DNS servers.
Can it help you with your restricted network ?

cbenhagen, 2017-06-20:
As I mentioned other DNS services are blocked on that network. So that would not work.

gregdel, 2017-06-20:
You can not do any DNS queries ? How does the OTB behave on this kind of networks ? (I've never tested that myself)

Can you connect to our shadowsocks proxy ? Do you have any logs ?
You can try to change the default behaviour of the init file by setting `disable_sni` to `false`, this is used to bypass the Chinese firewall... This might work, here's the line to change:
https://github.com/ovh/overthebox-feeds/blob/master/shadowsocks-libev/files/shadowsocks.init#L471

cbenhagen, 2017-06-21:
On that network all outgoing ports other than 80 and 443 were blocked. DNS queries could only be done to the locally provided DNS. That scenario should be pretty easy to reproduce.

Sorry I can't test this anymore. I canceled my subscription after I did not even get any answer to my repeated requests to the OVH support. I sent a more detailed report with logs through the support system. I just logged in there to get you a copy but unfortunately the tickets seem to have been deleted.

gregdel, 2017-06-21:
Thanks for the reply, we're focusing our work on making the product work for people who can manage their own network. It hasn't been designed to work on restricted network, that's why we didn't tested that use case.
We might do it later but it is not the current priority, thanks again for your feedback.

closed 2017-06-21 by gregdel