ovh/overthebox-feeds #346: How do you setup a vtun for MPTCP

slenpree, 2017-03-27:
Hi,

I have two WAN paths configured on a remote-side over-the-box load balancing nicely on a per-destination basis but the tun0 is down and I can't see where to configure the IP address of the other side of the tunnel in order to complete the MWAN setup and and get MPTCP running, I suspect this is also possible with a transparent SOCKs proxy but can't see how to correctly configure that either. I have a hub-side over-the-box with a public IP in a data-centre as well doing nothing at the moment, the idea would be to use this for the fast aggregated internet connection and terminate the remote-side node (only the remote-side has multiple WAN paths that need load balancing).

If anyone an point me in the right direction or towards any documentation that would be much appreciated. I haven't managed to find anything yet by searching the web.

Kind Regards,
Jonathan

gregdel, 2017-06-20:
Sorry for the late reply, do you still have issues finding the right configuration ?

slenpree, 2017-06-20:
In the end I'm running MPTCP patched Debian on both ends using openVPN and a proxy.

gregdel, 2017-06-20:
Yep, the web is not yet ready for MPTCP... That's why we're providing a proxy service to our customers.
Do you still use this project or do you have your custom setup on the "client" side ?

slenpree, 2017-06-20:
Using a custom setup at the moment on both sides for this particular use case.  Will continue to watch the OVH project and may revisit in future.

gregdel, 2017-06-20:
Thanks for opening this issue, sorry we don't answer your problems for now.

closed 2017-06-20 by gregdel