ovh/overthebox-feeds #489: daemon.err uhttpd[2025]: cut: standard output: Broken pipe

gregdel, 2017-06-30:
In luci, with the page `cgi-bin/luci//admin/network/network` open

gregdel, 2017-06-30:
```root@OverTheBox:/# mwan3 status
Interface status:
 interface tun0 is online (tracking active)
 interface if1 is online (tracking active)
 interface if2 is online (tracking active)

Policy balanced:
 if2 (50%)
 if1 (50%)

Policy balanced_tuns:
 tun0 (100%)

Policy dns_p_1:
cut: standard output: Broken pipe
 if2 (50%)
 if1 (50%)
```

gregdel, 2017-06-30:
```root@OverTheBox:/# mwan3 policies
Policy balanced:
cut: standard output: Broken pipe
 if2 (50%)
 if1 (50%)
```

sduponch, 2017-06-30:
root@OpenWrt:~# uci show mwan3.dns_p_1 ?

gregdel, 2017-06-30:
https://github.com/ovh/overthebox-feeds/blob/master/mwan3otb/files/usr/sbin/mwan3#L117-L141

[image]

gregdel, 2017-06-30:
@sduponch 

```
root@OverTheBox:/# uci show mwan3.dns_p_1
mwan3.dns_p_1=policy
mwan3.dns_p_1.use_member='if1_m1_w1' 'if2_m1_w1' 'tun0_m2_w1'
mwan3.dns_p_1.last_resort='default'
mwan3.dns_p_1.generated='1'
```

sduponch, 2017-06-30:
already seen "arithmetic error" here when use_member contains the same member twice, but this is the first time for this broken pipe.

here members looks good, do balancing between if1 and if2 and tun0 as last hope

closed 2017-07-20 by gregdel