ovh/overthebox-feeds #631: No MPTCP with version of 3rd august

epalzeolithe, 2017-08-05:
Hello

i used the build.sh script to build automatically the OTB image,

then i put it into a virtualbox.

Unfortunately, it seems to be a bug because, there is no "load balacing menu", and when i add the second connection through DHCP automatic detection, it doesnt add the 2 connections. No MPTCP zone in the firewall either.

Did i do something wrong or as it's the last build, it has this kind of bug ? Thanks in advance for the information

angt, 2017-08-06:
Hi,
You just have to check if the multipath option is set to "on" for your interface, everything else should be automatic.
We do not rely on mwan3 anymore and we have removed the mptcp zone for simplicity.
Also, you can download the last release here: http://downloads.overthebox.net/ovh/overthebox-feeds/master/targets/x86/64/

epalzeolithe, 2017-08-06:
Hello

Thank you for your Quick reponse !

I used the last build ;)

I ve understood that by default the lan access is at 192.168.100.1

Each Time i use the Wizard , everything works, the traffic goes Weil through the new gateway... until it asks me to disable DHCP on the first gateway. Each Time, whatever i do, the IF becomes offline and then it s broken. Reboot : same issue : the interface stays offline :(

epalzeolithe, 2017-08-06:
Fixed :)

I found a workaround

I just have to sleep the whole machine then wake up, and the connections are live again

closed 2017-08-06 by epalzeolithe