ovh/overthebox-feeds #632: Feature : TTL overriding

epalzeolithe, 2017-08-06:
Hi,

it's possible to override the TTL of  each packets when going through the gateway, to a default TTL ?

Thanks in advance :)

gregdel, 2017-08-07:
Hi, you can do this with the `iptables` command.
You should do it in the mangle table / postrouting chain.
e.g.
```
iptables -t mangle -I POSTROUTING -d 8.8.8.8 -p tcp -j TTL --ttl-set 1
```

However, according to the man (iptables-extensions):
> Setting or incrementing the TTL field can potentially be very dangerous, so it should be avoided at any cost.

You can add custom rules in the web interface under firewall > custom rules.

gregdel, 2017-08-21:
Any news ? Did you try the solution I gave you ?

epalzeolithe, 2017-08-21:
Hello, not yet, i had to fully reinstall my OTB, and changed the whole topology of my LAN,

i will :)

closed 2017-11-18 by gregdel

epalzeolithe, 2017-12-29:
Hello

i did it on my otb

the command iptables, is it persistent or do i need to run it at each start ?