ovh/overthebox-feeds #638: Firewall restarts on ifupdate

PouuleT, 2017-08-10:
` user.notice firewall: Reloading firewall due to ifupdate of if1 (if1)`

The ifupdate seems to be triggered from the ubus call set_data of /usr/share/otb/post-tracking.d/get-ip-asn

Packets could be loss while this is happening, we need to find a proper way to avoir reloading the firewall or understand exaclty why the ifupdate hotplug is triggered

It seems that if the data doesn't change, the trigger is not runned, so we need to check what exactly is changing in the data we set ( or in the data that is stored )
It seems to be reload way too often to be legit changes : 

```Thu Aug 10 11:53:19 2017 user.notice firewall: Reloading firewall due to ifupdate of if1 (if1)
Thu Aug 10 11:53:21 2017 user.notice firewall: Reloading firewall due to ifupdate of if1 (if1)
Thu Aug 10 11:53:47 2017 user.notice firewall: Reloading firewall due to ifupdate of if2 (if2)
Thu Aug 10 11:53:49 2017 user.notice firewall: Reloading firewall due to ifupdate of if2 (if2)
Thu Aug 10 11:55:38 2017 user.notice firewall: Reloading firewall due to ifupdate of if2 (if2)
Thu Aug 10 11:55:40 2017 user.notice firewall: Reloading firewall due to ifupdate of if2 (if2)
Thu Aug 10 11:58:59 2017 user.notice firewall: Reloading firewall due to ifupdate of if1 (if1)
Thu Aug 10 11:59:02 2017 user.notice firewall: Reloading firewall due to ifupdate of if1 (if1)
Thu Aug 10 11:59:47 2017 user.notice firewall: Reloading firewall due to ifupdate of if2 (if2)
Thu Aug 10 11:59:49 2017 user.notice firewall: Reloading firewall due to ifupdate of if2 (if2)
Thu Aug 10 12:00:13 2017 user.notice firewall: Reloading firewall due to ifupdate of if1 (if1)
Thu Aug 10 12:00:15 2017 user.notice firewall: Reloading firewall due to ifupdate of if1 (if1)
```

ping @Tikiwinkie

PouuleT, 2017-08-10:
It seems linked to when iptoasn doesn't respond on time
 -> We sent empty/invalid values with set_data that make the firewall reload

Maybe we should not do the set_data if iptoasn doesn't respond ?

Tikiwinkie, 2017-08-10:
Better adapt set_data content, depending on which informations we could retrieve

PouuleT, 2017-08-23:
Fixed by #643

closed 2017-08-23 by PouuleT

gregdel, 2017-09-11:
Linked to #707