ovh/overthebox-feeds #731: Add iPhone tethering capabilities

gregdel, 2017-09-14:
This packages must be installed: 
* kmod-usb-net-ipheth 
* libimobiledevice 
* usbmuxd

**usbmuxd must be started**

sduponch, 2017-09-14:
Maybe we can add a hotplug.d/usb script to start usbmuxd when we detect apple devices.

cf :
```
#!/bin/sh
BINARY="/sbin/wifi up"
RTL8188SU_PRODID="bda/8171/200"
if [ "${PRODUCT}" = "${RTL8188SU_PRODID}" ]; then
     if [ "${ACTION}" = "add" ]; then
         ${BINARY}
     fi
fi
```

gregdel, 2017-09-15:
That's what usb-modeswitch does:

```
root@glenn:/etc/hotplug.d/usb# cat 20-usb_mode
/etc/init.d/usbmode start
```
```
root@glenn:/etc/hotplug.d/usb# cat /etc/init.d/usbmode
#!/bin/sh /etc/rc.common

START=20
USE_PROCD=1

start_service()
{
	procd_open_instance
	procd_set_param command "/sbin/usbmode" -s
	procd_close_instance
}
```

closed 2019-02-26 by angt