ovh/overthebox-feeds #760: Can I use otb with my own router and shadowsocks/glorytun server?

trungpham, 2017-10-03:
Should I cherrypick the packages I need or deploy the entire otb stack as is then change the shadowsocks and glorytun server address? Thanks.

debiansid, 2017-10-04:
How did you activate your otb?

trungpham, 2017-10-04:
@debiansid that’s exactly the issue. I won’t be able to. So I will need a customized version of OTB where I only need to enter the shadowsocks and glorytun servers’ addresses.

debiansid, 2017-10-04:
@trungpham I have same issue. I was able to activate it on old version. For now , I thought ovh might consider to provide activation only service , then we can use our own shadowsocks and glorytun server.

trungpham, 2017-10-04:
Yah. Looks like we need our own fork of OTB. I got as far as patching the LEDE kernel to support MPTCP.

debiansid, 2017-10-04:
1.support modem in bridge mode 

2.awesome ovh-luci-app

I need these two.

gregdel, 2017-10-04:
Hey guys,

We won't help you to clone our infrastructure, it's what we're selling and it might change in the future. The device part however is open source and you can update / fork / do whatever you want with it :)

@trungpham, I'm not sure why you needed to fork lede as your version already includes MPTCP.

closed 2017-10-04 by angt