ovh/overthebox-feeds #831: Mac addresses not well configured for custom interfaces

gregdel, 2017-11-21:
The macaddr configuration is on the cifX instead of cifX_dev, so it won't work:
```
root@OverTheBox:~# uci show network.cif1.macaddr
network.cif1.macaddr='0A:8F:B9:E2:F4:BA'
root@OverTheBox:~# ip link show cif2
23: cif2@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc cake state UNKNOWN mode DEFAULT group default qlen 1
    link/ether 5e:3e:e8:74:02:e7 brd ff:ff:ff:ff:ff:ff
```

Fixed my moving the config to cifX_dev:
```
root@OverTheBox:~# uci show network.cif1_dev.macaddr
network.cif1_dev.macaddr='0A:8F:B9:E2:F4:BA'
root@OverTheBox:~# ip link show cif1
26: cif1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1
    link/ether 0a:8f:b9:e2:f4:ba brd ff:ff:ff:ff:ff:ff
```

gregdel, 2017-11-21:
The macaddr is chosen randomly if none is specified. If the client change it in the web interface, it is configured in the wrong uci section.

angt, 2018-08-21:
This was not a real issue, macaddr was not correct (and useless) on cifX.

closed 2018-08-21 by angt