ovh/overthebox-feeds #839: Tethering iPhone not working on 0.6.X ??

gregdel, 2017-11-22:
Some users claims that the iPhone tethering is not working

gregdel, 2017-12-06:
Following this guide:
https://docs.ovh.com/fr/overthebox/tethering-iphone-et-android/#pour-un-telephone-iphone

I've launched `usbmuxd` manually and updated the line in the `/etc/rc.local` file as described in the doc.

If the iPhone is plugged, an interface is created `eth1` in my case. I've setup the eth1 configuration from the web ui and it works !

gregdel, 2017-12-06:
```
[10184.038144] ------------[ cut here ]------------                                                                                                                                         
[10184.044025] WARNING: CPU: 1 PID: 0 at net/sched/sch_generic.c:306 dev_watchdog+0x1e8/0x200()                                                                                             
[10184.049914] NETDEV WATCHDOG: eth1 (ipheth): transmit queue 0 timed out                                                                                                                   
[10184.055814] Modules linked in: qcserial option ipw iptable_nat cdc_mbim usb_wwan usb_serial_simple ti_usb_3410_5052 sr9700 smsc95xx sierra_net sierra rndis_host qmi_wwan pppoe          
ppp_async pl2303 oti6858 nf_nat_pptp nf_nat_ipv4 nf_nat_amanda nf_conntrack_pptp nf_conntrack_ipv6 nf_conntrack_ipv4 nf_conntrack_amanda mos7720 mct_u232 mcs7830 keyspan kalmia ipt_REJECT 
ipt_MASQUERADE huawei_cdc_ncm garmin_gps ftdi_sio dm9601 cypress_m8 cp210x ch341 cdc_subset cdc_ncm cWed Dec  6 13:55:38 2017 kern.warn kernel: [10184.147851] CPU: 1 PID: 0 Comm: swapper/ 
1 Not tainted 4.4.93 #0                                                                                                                                                                     
[10184.157825] Hardware name: OVH SAS OverTheBox V2A/Cherry Trail CR, BIOS 5.11 02/27/2017                                                                                                  
[10184.167888]  0000000000000000 ffffffff81219252 ffff88007ac83e60 ffffffff816d59f7                                                                                                         
[10184.178001]  ffffffff81071d83 0000000000000000 ffff88007ac83eb0 0000000000000001                                                                                                         
[10184.188250]  0000000000000001 ffff88000006e3e0 ffffffff81071df7 ffffffff816d8ed0                                                                                                         
[10184.198478] Call Trace:                                                                                                                                                                  
[10184.208569]  <IRQ>  [<ffffffff81219252>] ? dump_stack+0x5c/0x7a                                                                                                                          
[10184.218800]  [<ffffffff81071d83>] ? warn_slowpath_common+0x73/0xa0                                                                                                                       
[10184.229044]  [<ffffffff81071df7>] ? warn_slowpath_fmt+0x47/0x50                                                                                                                          
[10184.239213]  [<ffffffff810b2746>] ? mod_timer_pinned+0xa6/0xe0                                                                                                                           
[10184.249347]  [<ffffffff81449788>] ? dev_watchdog+0x1e8/0x200                                                                                                                             
[10184.259493]  [<ffffffff814495a0>] ? qdisc_rcu_free+0x40/0x40                                                                                                                             
[10184.269558]  [<ffffffff810b2981>] ? call_timer_fn.isra.4+0x11/0x70                                                                                                                       
[10184.279546]  [<ffffffff810b2b10>] ? run_timer_softirq+0x130/0x1f0                                                                                                                        
[10184.289533]  [<ffffffff810750f6>] ? __do_softirq+0xc6/0x1d0                                                                                                                              
[10184.299563]  [<ffffffff8107531f>] ? irq_exit+0x6f/0x80                                                                                                                                   
[10184.309528]  [<ffffffff81038d13>] ? smp_trace_apic_timer_interrupt+0x63/0x90                                                                                                             
[10184.319517]  [<ffffffff8152903f>] ? apic_timer_interrupt+0x7f/0x90                                                                                                                       
[10184.329496]  <EOI>  [<ffffffff813d919d>] ? cpuidle_enter_state+0x12d/0x1e0                                                                                                               
[10184.339525]  [<ffffffff813d9196>] ? cpuidle_enter_state+0x126/0x1e0                                                                                                                      
[10184.349481]  [<ffffffff8109ef04>] ? cpu_startup_entry+0x1f4/0x250                                                                                                                        
[10184.359425]  [<ffffffff8103725d>] ? start_secondary+0x12d/0x140                                                                                                                          
[10184.369341] ---[ end trace 87fd630114f5c06d ]--- 
```
I got this stack trace but the device did not reboot and it worked afterwards

gregdel, 2017-12-06:
`usbmuxd` is not launched after reboot, i launched the command manually and it worked:
[image]

gregdel, 2017-12-06:
The connection does not work if I unplug / plug the iPhone.
To make it work I had to kill `usbmuxd` and launch it again.

CelineDesign, 2020-08-28:
Did the same process as you but doesn't work :
I trust the connection on the iPhone screen, iPhone is connected, online, then few second later offline in the OTB panel.
i always had the same error in Kernel log : ipheth 2-2:4.2: ipheth_tx_timeout: TX timeout
and the iPhone interface is offline

If i deactivate multipath iPhone stay online on the OTB admin panel.

I found a possible fix here : 
https://forum.openwrt.org/t/iphone-tethering-ipheth-kernel-driver-fails/16973
but i don't know how to upgrade the OTB Kernel…

Please help

AxelDum, 2024-04-25:
Closing fixed on v0.8

closed 2024-04-25 by AxelDum