ovh/overthebox-feeds #843: QoS based on multiple domain names

gregdel, 2017-11-24:
We might be able to resolve all the domain names with the same QoS priority, resolve them and group them in a single ipset for dscp matching.
This would allow flexible QoS configuration en efficient matching.

YannickBiet, 2017-11-24:
Functionnal scenario is to be able to set as low priority (or limit the BW) for windows update / dropbox / google drive / one drive / ....

sduponch, 2017-11-24:
The use of dnsmasq ipset allow you to add resolved address matching a patern to an ipset.
ipset=/netflix/,cs1

gregdel, 2017-11-24:
The ipset needs to be created first:
```
root@OverTheBox:~# ipset create yolo hash:ip family inet
```

Then you can add this conf in dnsmasq:
```
config dnsmasq
        .....
        list ipset '/google.com/yolo' 
```

After the first dns query to `google.com`:
```
root@OverTheBox:~# ipset list yolo
Name: yolo
Type: hash:ip
Revision: 4
Header: family inet hashsize 1024 maxelem 65536
Size in memory: 144
References: 0
Members:
216.58.209.238
```

YannickBiet, 2017-11-24:
Hi Sebastien,

I think that the FR is to be able to do from the UI of the OTB.
Tell me if I am wrong, but for me your answer needs to be connected in SSH and know howto/where edit the right config file ? am i wrong ?

sduponch, 2017-11-24:
@YannickBiet yes this an experimental way to do it, we can integrate this in the UI later.

Main drawback is that dnsmasq's ipset are located on the device that only handle the upload side of the QoS, so to use this feature with the download side we need a way to push resolved address to the service side via a dnsmasq patch or whatever.

@gregdel Maybe the QoS service could pre-create an ipset for each QoS queues ?

gregdel, 2017-11-24:
I'm not sure using dnsmasq to fill the ipset is a good idea, this will create a hard dependency between two unrelated packages: dnsmasq(dns/dhcp) and otb-qos(qos). But this is my personal opinion.

closed 2024-06-10 by AxelDum