ovh/overthebox-feeds #855: Aggregated links, download rate not stable

epalzeolithe, 2017-12-09:
Hi

i've got 2 xDSL links totally stable.

With the previous version of OTB (0.5 out of the box), aggregated links were working well and stable.

SInce i upgraded to last firmware (even the last build Powered by LuCI 17.11.03 Branch (v1.1) / otb Reboot v3.3 v0.6.17-12-gaf569fb)

the output is not stable. Every 5 seconds, it comes close to zero, then goes back to maximum speed. I lose the benefit of the solution as the average output is lower than the slowest link.

I tried to setup traffic control limits if ever it was a congestion of packets. Nothing better.

Do you know what's happening, because i'm going to resign the service. I d'ont understand, it was working well with previous firmware ouf of the box

gregdel, 2017-12-09:
Can you give me your device id so I can check your device configuration ?
Do you authorize me to remotely connect to your device ?

FYI we also have an active mailing list for the product: https://www.ovh.com/fr/support/mailing.xml

epalzeolithe, 2017-12-09:
It's this NUC Intel DE3815TYKHE

How do you want to connect remotely ?

epalzeolithe, 2017-12-09:
334455d1-6ba4-4e08-a170-37cecb4711a1

gregdel, 2017-12-09:
There is a feature you can find in the OVH manager to create a remote access to your device, the support team can ask for a remote access and you can validate it through the manager. Only @PouuleT and I can auto-validate these to help customers.

I've looked at your network configuration, there a lot of unused interfaces and some configuration issues VLAN over macvlan for exemple.
I've backed up your previous configuration and fixed your network configuration.

Tell me if you see an improvement in the aggregation.
You can always restore your old configuration using the API: https://api.ovh.com/console/#/overTheBox/%7BserviceName%7D/device/restoreBackup#POST

epalzeolithe, 2017-12-10:
Thank you very much you fixed it :) :) :) 👍 

I followed an old tutorial where for VLAN we had to previously set un unmanged interface, and my config was so dirty :(

THanks again !

closed 2017-12-10 by epalzeolithe

gregdel, 2017-12-10:
Glad I could help :)