ovh/overthebox-feeds #863: TTL persistent

epalzeolithe, 2017-12-30:
Hello

I need to change on the fly the TTL on each outgoing packet through each WAN (not the packets going through the tunnel)

Is il the right command for the WANs ?

iptables -t mangle -I POSTROUTING -d 8.8.8.8 -p tcp -j TTL --ttl-set 65

And is it persistent after reboot?

gregdel, 2017-12-30:
iptables is not persistent, you can add the rules manually in `/etc/firewall.user` or via the web interface "Firewall > Custom Rules". If your custom rules are in this file, they will get applied on the next reboot.

What's your use case for changing the TTL ?

epalzeolithe, 2017-12-30:
tethering cloaking

epalzeolithe, 2017-12-31:
Well, iptables change seems to be persistent

i use this command instead for each VLAN

iptables -t mangle -A PREROUTING -i eth0.3 -j TTL --ttl-set 65

gregdel, 2018-01-02:
Does this iptables command fixes your "issue" ? :)

closed 2018-01-03 by epalzeolithe