ovh/overthebox-feeds #869: Issue when installing image

epalzeolithe, 2018-01-03:
Hello

i'm trying to install OTB on a Z8350 II intel box. This box runs fine Windows and Linux.

I followed this tutorial : https://docs.ovh.com/fr/overthebox/installer-limage-overthebox-sur-votre-materiel/

But when it boots, it's hanging on : Waiting for root device PARTUUID=48a1b25e-02...

I dont have options in bios. The machine starts strait away a EFI shell

Thanks in advance :)

epalzeolithe, 2018-01-03:
I captured the shell, here's what i see 

first it uses the startup.nsh script

ithe efi script scan the boot entries

then select fs0:

and launch fs0:\efi\boot\boot.nsh

it launchs : linux.efi root=PARTUUID=48a1b25e-02 rootfstype=squashfs rootwait console=tty0

then it launchs OTB but hangs on Waiting for root device  PARTUUID=48a1b25e-02

angt, 2018-01-03:
Hi,
No idea :(
But you can try to overwrite the root option in the file boot.nsh (you should have an edit command).
You can also check the PARTUUID with the command `blkid`.
Regards

epalzeolithe, 2018-01-03:
I did a blkid

it's not exactly the same, i've got 48a1b25e-01 instead of 48a1b25e-02

But after editing and put 01 instead of 02, same issue :(

epalzeolithe, 2018-01-03:
I put otb on an usb key, and it works.

But usb key is slower. Is it important or not for OTB ? Everything is in RAM ?

epalzeolithe, 2018-01-03:
Ok it dont work, i need to manually enter the shell to launch manually boot.nsh on usb. Anythinh else fails.

So i installed ubuntu aside OTB.
OTB squashfs image is still on /dev/mmcblk0p2

I'd like to add a grub entry to boot the squashfs OTB, does anyone what i have to put in grub to boot OTB ?

epalzeolithe, 2018-01-03:
Ok, i managed to boot the squashfs image from grub

Same bug > it can't find the PARTUUID... "Waiting for root device" :( :( :(

epalzeolithe, 2018-01-03:
Well fixed.
1) the provided kernel stalls on boot, whatever booting method i use (linux.efi or vmlinuz through grub)
2) i put the OTB image on usb key
3) it's buggy also through linux.efi. But through grub it works, using root=/dev/sda

pfff, quite to hard to workaround

now i've got 4 cores running for OTB, on a box less than 80$

angt, 2018-01-04:
Nice :)
I close this issue as we don't want to officially support this hw.

closed 2018-01-04 by angt

sduponch, 2018-01-08:
This look like a missing kernel module that handle your drive controller, this could append if your device does not use an embed eMMC chip or have an exotic SATA controller.