ovh/overthebox-feeds #889: eMMC driver

epalzeolithe, 2018-02-10:
Hello

i'm trying to boot from an eMMC module,

it seems that a config or module is missing.

I can boot ubuntu from the eMMC module. How can i identify which module is missing to insert it into overthebox build ? Thanks

gregdel, 2018-02-13:
Hello,
What kind of hardware are you using ? Did you manage to install overthebox on it ?

epalzeolithe, 2018-02-13:
Hello

that's an Z83II with atom Z8350 and embedded MMC. It's an intel nuc.

Yes overthebox is working great from an USB key. But the overthebox image can't find the emmc disk during boot. Ubuntu does. Both are using sdhci and mmc modules.

 I compiled the last dev image of overthebox. I checked the menuconfig of the kernel, thoses modules are activated as built-in. I've also checked the kernel flag config_x86_intel_lpss because i saw it's often needed for emmc proprer detection. No luck, the emmc drive is not seen on boot.

I dumped with lsmod to see differences between ubuntu and the overthebox. Perhaps mmc_spi driver could be the difference but i'm not sure...

Thanks in advance for any help :)

angt, 2019-02-26:
Timeout

closed 2019-02-26 by angt