ovh/overthebox-feeds #1104: Glorytun do not set ppoe interface

BenoitAnastay, 2019-12-16:
My PPoE interface is not shoing up in glorytun despite the fact everything is configured correctly

I made a script to try to debug it 
```
for iface in $(uci -q get firewall.wan.network); do

network_get_ipaddr ipaddr "$iface"
network_get_gateway gateway "$iface"
network_get_subnet subnet "$iface"

multipath=$(uci -q get "network.$iface.multipath")
upload=$(uci -q get "network.$iface.upload")
download=$(uci -q get "network.$iface.download")
table=$(ip -o route get from "$ipaddr" to 8.8.8.8 2>&1 )

if [ ! $ipaddr == "" ] ; then
echo "$iface"
echo "ipaddr : $ipaddr"
echo "gateway: $gateway"
echo "subnet: $subnet"
echo "multipath: $multipath"
echo "download: $download"
echo "upload: $upload"
echo "table: $table"
glorytun path "$ipaddr" up rate rx "$download" tx "$upload"
echo "---"
fi
done
glorytun path
```

The result is 
```
free
ipaddr : 78.219.164.2
gateway: 78.219.164.254
subnet: 78.219.164.2/24
multipath: on
download: 16Mbits
upload: 1Mbits
table: 8.8.8.8 from 78.219.164.2 via 78.219.164.254 dev eth1.501 table 201 uid 0 \    cache
---
sfr
ipaddr : 77.204.219.206
gateway: 77.204.218.1
subnet: 77.204.219.206/23
multipath: on
download: 100Mbits
upload: 5Mbits
table: 8.8.8.8 from 77.204.219.206 via 77.204.218.1 dev eth1.503 table 202 uid 0 \    cache
---
ovh
ipaddr : 109.190.21.56
gateway: 145.239.153.1
subnet: 109.190.21.56/32
multipath: on
download: 12Mbits
upload: 800Kbits
table: 8.8.8.8 from 109.190.21.56 via 145.239.153.1 dev pppoe-ovh table 200 uid 0 \    cache
---
path UP
  status:  OK
  bind:    78.219.164.2 port 5006
  public:  78.219.164.2 port 5006
  peer:    109.190.219.130 port 5006
  mtu:     1448 bytes
  rtt:     26.705 ms
  rttvar:  0.203 ms
  tx:
    rate:  125000 bytes/sec
    loss:  0 percent
    total: 40853 packets
  rx:
    rate:  2000000 bytes/sec
    loss:  0 percent
    total: 23546 packets
path UP
  status:  OK
  bind:    77.204.219.206 port 5006
  public:  77.204.219.206 port 5006
  peer:    109.190.219.130 port 5006
  mtu:     1448 bytes
  rtt:     26.365 ms
  rttvar:  0.415 ms
  tx:
    rate:  625000 bytes/sec
    loss:  0 percent
    total: 88728 packets
  rx:
    rate:  12500000 bytes/sec
    loss:  0 percent
    total: 69240 packets
path UP
  status:  DEGRADED
  bind:    109.190.21.56 port 5006
  public:  - port 0
  peer:    109.190.219.130 port 5006
  mtu:     1302 bytes
  rtt:     0.000 ms
  rttvar:  0.000 ms
  tx:
    rate:  100000 bytes/sec
    loss:  0 percent
    total: 1 packets
  rx:
    rate:  1500000 bytes/sec
    loss:  0 percent
    total: 0 packets
```
As you can see it doens't get the interface settings, and then the path is removed from glorytun

The ifconfig result 
```
pppoe-ovh Link encap:Point-to-Point Protocol
          inet addr:109.190.21.56  P-t-P:145.239.153.1  Mask:255.255.255.255
          inet6 addr: fe80::bcda:c9a2:3873:8796/10 Scope:Link
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1492  Metric:1
          RX packets:9032 errors:0 dropped:0 overruns:0 frame:0
          TX packets:11961 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:3
          RX bytes:5205786 (4.9 MiB)  TX bytes:1594604 (1.5 MiB)

eth1.502  Link encap:Ethernet  HWaddr 80:EE:73:95:C8:30
          inet6 addr: fe80::82ee:73ff:fe95:c830/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:9329 errors:0 dropped:0 overruns:0 frame:0
          TX packets:12039 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:5299010 (5.0 MiB)  TX bytes:1861134 (1.7 MiB)
```

I guess there is an issue with the fact that the interface is named pppoe-ovh but don't know why
The interface itself is working properly 

Because it's a glorytun related issue I'm not sure if it's the right place to post this issue, but I didn't tried outside OTB to check which of both is causing this

angt, 2019-12-16:
Hi,
Can you confirm udp works with this modem ?

BenoitAnastay, 2019-12-16:
The iperf result, UDP seem to work properly 

```
root@OverTheBox:~# iperf3 -c ping.online.net -u -B 109.190.21.56
Connecting to host ping.online.net, port 5201
[  5] local 109.190.21.56 port 57054 connected to 62.210.18.40 port 5201
[ ID] Interval           Transfer     Bitrate         Total Datagrams
[  5]   0.00-1.00   sec   128 KBytes  1.05 Mbits/sec  91
[  5]   1.00-2.00   sec   128 KBytes  1.05 Mbits/sec  91
[  5]   2.00-3.00   sec   128 KBytes  1.05 Mbits/sec  91
[  5]   3.00-4.00   sec   129 KBytes  1.06 Mbits/sec  92
[  5]   4.00-5.00   sec   128 KBytes  1.05 Mbits/sec  91
[  5]   5.00-6.00   sec   128 KBytes  1.05 Mbits/sec  91
[  5]   6.00-7.00   sec   128 KBytes  1.05 Mbits/sec  91
[  5]   7.00-8.00   sec   128 KBytes  1.05 Mbits/sec  91
[  5]   8.00-9.00   sec   128 KBytes  1.05 Mbits/sec  91
[  5]   9.00-10.00  sec   128 KBytes  1.05 Mbits/sec  91
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Jitter    Lost/Total Datagrams
[  5]   0.00-10.00  sec  1.25 MBytes  1.05 Mbits/sec  0.000 ms  0/911 (0%)  sender
[  5]   0.00-10.00  sec  1.06 MBytes   888 Kbits/sec  2.367 ms  0/771 (0%)  receiver

root@OverTheBox:~# iperf3  -s -B 109.190.21.56 // [root@ns392XXX ~]# iperf3 -c 109.190.21.56 -u
-----------------------------------------------------------
Server listening on 5201
-----------------------------------------------------------
Accepted connection from 176.31.XXX.XXX, port 58080
[  5] local 109.190.21.56 port 5201 connected to 176.31.XXX.XXX port 42208
[ ID] Interval           Transfer     Bitrate         Jitter    Lost/Total Datagrams
[  5]   0.00-1.00   sec   115 KBytes   945 Kbits/sec  1.600 ms  0/82 (0%)
[  5]   1.00-2.00   sec   128 KBytes  1.05 Mbits/sec  1.596 ms  0/91 (0%)
[  5]   2.00-3.00   sec   128 KBytes  1.05 Mbits/sec  1.603 ms  0/91 (0%)
[  5]   3.00-4.00   sec   129 KBytes  1.06 Mbits/sec  1.536 ms  0/92 (0%)
[  5]   4.00-5.00   sec   128 KBytes  1.05 Mbits/sec  1.544 ms  0/91 (0%)
[  5]   5.00-6.00   sec   128 KBytes  1.05 Mbits/sec  1.577 ms  0/91 (0%)
[  5]   6.00-7.00   sec   128 KBytes  1.05 Mbits/sec  1.581 ms  0/91 (0%)
[  5]   7.00-8.00   sec   128 KBytes  1.05 Mbits/sec  1.604 ms  0/91 (0%)
[  5]   8.00-9.00   sec   128 KBytes  1.05 Mbits/sec  1.611 ms  0/91 (0%)
[  5]   9.00-10.00  sec   128 KBytes  1.05 Mbits/sec  1.591 ms  0/91 (0%)
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Jitter    Lost/Total Datagrams
[  5]   0.00-10.07  sec  1.24 MBytes  1.03 Mbits/sec  1.591 ms  0/902 (0%)  receiver

root@OverTheBox:~# iperf3  -s -B 109.190.21.56 // [root@ns392XXX ~]# iperf3 -c 109.190.21.56 -u -R
-----------------------------------------------------------
Server listening on 5201
-----------------------------------------------------------
Accepted connection from 176.31.XXX.XXX, port 58086
[  5] local 109.190.21.56 port 5201 connected to 176.31.XXX.XXX port 37289
[ ID] Interval           Transfer     Bitrate         Total Datagrams
[  5]   0.00-1.00   sec   128 KBytes  1.05 Mbits/sec  91
[  5]   1.00-2.00   sec   128 KBytes  1.05 Mbits/sec  91
[  5]   2.00-3.00   sec   128 KBytes  1.05 Mbits/sec  91
[  5]   3.00-4.00   sec   129 KBytes  1.06 Mbits/sec  92
[  5]   4.00-5.00   sec   128 KBytes  1.05 Mbits/sec  91
[  5]   5.00-6.00   sec   128 KBytes  1.05 Mbits/sec  91
[  5]   6.00-7.00   sec   128 KBytes  1.05 Mbits/sec  91
[  5]   7.00-8.00   sec   128 KBytes  1.05 Mbits/sec  91
[  5]   8.00-9.00   sec   128 KBytes  1.05 Mbits/sec  91
[  5]   9.00-10.00  sec   128 KBytes  1.05 Mbits/sec  91
[  5]  10.00-10.08  sec  9.84 KBytes   952 Kbits/sec  7
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Jitter    Lost/Total Datagrams
[  5]   0.00-10.08  sec  1.26 MBytes  1.05 Mbits/sec  0.000 ms  0/918 (0%)  sender
```

angt, 2019-12-20:
Thank you for your tests.
Why do you have a txqlen of 3 on your pppoe interface, do you use sqm ?
Can you confirm that packets with the correct source and destination ip come out of the tun interface?

BenoitAnastay, 2019-12-20:
> Why do you have a txqlen of 3 on your pppoe interface, do you use sqm ?
I don't know why, i'm just using the default configuration, I don't have SQM, the only change I made on the build is to compile all kmod but I didn't installed any because I didn't found the one I was looking for (rtsp nathelper) so it's a bare build with defaults setting, I have just added my interfaces and put PPPOE settings 

```
root@OverTheBox:~# tcpdump -i eth1.502 | grep 5006
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on eth1.502, link-type EN10MB (Ethernet), capture size 262144 bytes
15:10:34.809101 PPPoE  [ses 0x1] IP 56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: UDP, length 1386
15:10:34.855895 PPPoE  [ses 0x1] IP 130-219-190-109.isp.overthebox.ovh.5006 > 56-21-190-109.dsl.ovh.fr.5006: UDP, length 1386
15:10:34.856234 PPPoE  [ses 0x1] IP 56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: UDP, length 1386
15:10:35.547442 PPPoE  [ses 0x1] IP 130-219-190-109.isp.overthebox.ovh.5006 > 56-21-190-109.dsl.ovh.fr.5006: UDP, length 210
```

There is just those 4 packets exchanged when I run `glorytun path` then nothing

BenoitAnastay, 2019-12-20:
On a second run I get this 

```
root@OverTheBox:~# tcpdump -i eth1.502 -vv | grep 5006
tcpdump: listening on eth1.502, link-type EN10MB (Ethernet), capture size 262144 bytes
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 1386
    130-219-190-109.isp.overthebox.ovh.5006 > 56-21-190-109.dsl.ovh.fr.5006: [udp sum ok] UDP, length 1386
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 1386
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 62
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 90
    130-219-190-109.isp.overthebox.ovh.5006 > 56-21-190-109.dsl.ovh.fr.5006: [udp sum ok] UDP, length 1386
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 82
    130-219-190-109.isp.overthebox.ovh.5006 > 56-21-190-109.dsl.ovh.fr.5006: [udp sum ok] UDP, length 1386
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 74
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 74
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 74
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 74
    130-219-190-109.isp.overthebox.ovh.5006 > 56-21-190-109.dsl.ovh.fr.5006: [udp sum ok] UDP, length 1386
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 74
    130-219-190-109.isp.overthebox.ovh.5006 > 56-21-190-109.dsl.ovh.fr.5006: [udp sum ok] UDP, length 1386
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 74
    130-219-190-109.isp.overthebox.ovh.5006 > 56-21-190-109.dsl.ovh.fr.5006: [udp sum ok] UDP, length 1386
    130-219-190-109.isp.overthebox.ovh.5006 > 56-21-190-109.dsl.ovh.fr.5006: [udp sum ok] UDP, length 1386
    130-219-190-109.isp.overthebox.ovh.5006 > 56-21-190-109.dsl.ovh.fr.5006: [udp sum ok] UDP, length 1386
    130-219-190-109.isp.overthebox.ovh.5006 > 56-21-190-109.dsl.ovh.fr.5006: [udp sum ok] UDP, length 1386
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 62
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 82
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 82
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 74
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 82
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 82
    130-219-190-109.isp.overthebox.ovh.5006 > 56-21-190-109.dsl.ovh.fr.5006: [udp sum ok] UDP, length 1386
    130-219-190-109.isp.overthebox.ovh.5006 > 56-21-190-109.dsl.ovh.fr.5006: [udp sum ok] UDP, length 1386
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 62
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 62
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 82
    130-219-190-109.isp.overthebox.ovh.5006 > 56-21-190-109.dsl.ovh.fr.5006: [udp sum ok] UDP, length 1386
    130-219-190-109.isp.overthebox.ovh.5006 > 56-21-190-109.dsl.ovh.fr.5006: [udp sum ok] UDP, length 1386
    130-219-190-109.isp.overthebox.ovh.5006 > 56-21-190-109.dsl.ovh.fr.5006: [udp sum ok] UDP, length 1386
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 90
    130-219-190-109.isp.overthebox.ovh.5006 > 56-21-190-109.dsl.ovh.fr.5006: [udp sum ok] UDP, length 1386
    130-219-190-109.isp.overthebox.ovh.5006 > 56-21-190-109.dsl.ovh.fr.5006: [udp sum ok] UDP, length 1386
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 62
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 74
    130-219-190-109.isp.overthebox.ovh.5006 > 56-21-190-109.dsl.ovh.fr.5006: [udp sum ok] UDP, length 1386
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 62
    130-219-190-109.isp.overthebox.ovh.5006 > 56-21-190-109.dsl.ovh.fr.5006: [udp sum ok] UDP, length 1386
    130-219-190-109.isp.overthebox.ovh.5006 > 56-21-190-109.dsl.ovh.fr.5006: [udp sum ok] UDP, length 1386
    130-219-190-109.isp.overthebox.ovh.5006 > 56-21-190-109.dsl.ovh.fr.5006: [udp sum ok] UDP, length 1386
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 74
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 74
    130-219-190-109.isp.overthebox.ovh.5006 > 56-21-190-109.dsl.ovh.fr.5006: [udp sum ok] UDP, length 1386
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 90
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 82
    130-219-190-109.isp.overthebox.ovh.5006 > 56-21-190-109.dsl.ovh.fr.5006: [udp sum ok] UDP, length 1386
[... cut ...]
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 74
    130-219-190-109.isp.overthebox.ovh.5006 > 56-21-190-109.dsl.ovh.fr.5006: [udp sum ok] UDP, length 1386
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 74
    56-21-190-109.dsl.ovh.fr.5006 > 130-219-190-109.isp.overthebox.ovh.5006: [udp sum ok] UDP, length 74
    130-219-190-109.isp.overthebox.ovh.5006 > 56-21-190-109.dsl.ovh.fr.5006: [udp sum ok] UDP, length 98
```

angt, 2019-12-20:
Do you see the traffic coming back in the tun interface ?

BenoitAnastay, 2019-12-20:
Only IPV6 it seem 

```
root@OverTheBox:~# tcpdump -i tun0  -vv
tcpdump: listening on tun0, link-type RAW (Raw IP), capture size 262144 bytes
16:43:38.484445 IP (tos 0x0, ttl 64, id 3965, offset 0, flags [DF], proto UDP (17), length 117)
    10.166.179.2.54027 > 173.ip-167-114-241.eu.22837: [udp sum ok] UDP, length 89
16:43:38.664367 IP6 (flowlabel 0x9f4d5, hlim 64, next-header UDP (17) payload length: 44) 2001:41d0:fe89:5f00::1.55140 > dns1.isp.ovh.net.53: [udp sum ok] 50417+ A? cdn.optimizely.com. (36)
16:43:38.664401 IP6 (flowlabel 0x64ce1, hlim 64, next-header UDP (17) payload length: 44) 2001:41d0:fe89:5f00::1.55140 > dns2.isp.ovh.net.53: [udp sum ok] 50417+ A? cdn.optimizely.com. (36)
16:43:38.664423 IP6 (flowlabel 0x713c7, hlim 64, next-header UDP (17) payload length: 44) 2001:41d0:fe89:5f00::1.55140 > dns2.isp.ovh.net.53: [udp sum ok] 50417+ A? cdn.optimizely.com. (36)
16:43:39.550770 IP (tos 0x0, ttl 64, id 4043, offset 0, flags [DF], proto UDP (17), length 117)
    10.166.179.2.54027 > 173.ip-167-114-241.eu.22837: [udp sum ok] UDP, length 89
16:43:39.567703 IP6 (flowlabel 0x3be56, hlim 64, next-header UDP (17) payload length: 42) 2001:41d0:fe89:5f00::1.17365 > dns1.isp.ovh.net.53: [udp sum ok] 2175+ A? s3.amazonaws.com. (34)
16:43:39.567730 IP6 (flowlabel 0xf6603, hlim 64, next-header UDP (17) payload length: 42) 2001:41d0:fe89:5f00::1.17365 > dns2.isp.ovh.net.53: [udp sum ok] 2175+ A? s3.amazonaws.com. (34)
16:43:39.567746 IP6 (flowlabel 0x8252c, hlim 64, next-header UDP (17) payload length: 42) 2001:41d0:fe89:5f00::1.17365 > dns2.isp.ovh.net.53: [udp sum ok] 2175+ A? s3.amazonaws.com. (34)
16:43:39.973331 IP6 (flowlabel 0x6ae68, hlim 64, next-header ICMPv6 (58) payload length: 88) 2001:41d0:fe89:5f00::1 > par21s03-in-x0a.1e100.net: [icmp6 sum ok] ICMP6, destination unreachable, unreachable address 2001:41d0:fe89:5f04:8d0a:5eca:62fd:ea5e
16:43:39.973364 IP6 (flowlabel 0x6ae68, hlim 64, next-header ICMPv6 (58) payload length: 88) 2001:41d0:fe89:5f00::1 > par21s03-in-x0a.1e100.net: [icmp6 sum ok] ICMP6, destination unreachable, unreachable address 2001:41d0:fe89:5f04:8d0a:5eca:62fd:ea5e
16:43:39.973382 IP6 (flowlabel 0x86b87, hlim 64, next-header ICMPv6 (58) payload length: 88) 2001:41d0:fe89:5f00::1 > 2607:f8b0:4023::bc: [icmp6 sum ok] ICMP6, destination unreachable, unreachable address 2001:41d0:fe89:5f04:8d0a:5eca:62fd:ea5e
16:43:39.973396 IP6 (flowlabel 0x86b87, hlim 64, next-header ICMPv6 (58) payload length: 88) 2001:41d0:fe89:5f00::1 > 2607:f8b0:4023::bc: [icmp6 sum ok] ICMP6, destination unreachable, unreachable address 2001:41d0:fe89:5f04:8d0a:5eca:62fd:ea5e
16:43:39.973412 IP6 (flowlabel 0x6ae68, hlim 64, next-header ICMPv6 (58) payload length: 88) 2001:41d0:fe89:5f00::1 > par21s03-in-x0a.1e100.net: [icmp6 sum ok] ICMP6, destination unreachable, unreachable address 2001:41d0:fe89:5f04:8d0a:5eca:62fd:ea5e
16:43:39.973427 IP6 (flowlabel 0x6ae68, hlim 64, next-header ICMPv6 (58) payload length: 88) 2001:41d0:fe89:5f00::1 > par21s03-in-x0a.1e100.net: [icmp6 sum ok] ICMP6, destination unreachable, unreachable address 2001:41d0:fe89:5f04:8d0a:5eca:62fd:ea5e
16:43:39.973443 IP6 (flowlabel 0xfc55d, hlim 64, next-header ICMPv6 (58) payload length: 88) 2001:41d0:fe89:5f00::1 > par10s33-in-x04.1e100.net: [icmp6 sum ok] ICMP6, destination unreachable, unreachable address 2001:41d0:fe89:5f04:8d0a:5eca:62fd:ea5e
16:43:40.617262 IP (tos 0x0, ttl 64, id 4074, offset 0, flags [DF], proto UDP (17), length 117)
    10.166.179.2.54027 > 173.ip-167-114-241.eu.22837: [udp sum ok] UDP, length 89
16:43:41.682834 IP (tos 0x0, ttl 64, id 4083, offset 0, flags [DF], proto UDP (17), length 117)
    10.166.179.2.54027 > 173.ip-167-114-241.eu.22837: [udp sum ok] UDP, length 89
16:43:42.617488 IP6 (flowlabel 0x91f19, hlim 64, next-header UDP (17) payload length: 55) 2001:41d0:fe89:5f00::1.2384 > dns1.isp.ovh.net.53: [udp sum ok] 5990+ A? addons-pa.clients6.google.com. (47)
16:43:42.617520 IP6 (flowlabel 0x53a2d, hlim 64, next-header UDP (17) payload length: 55) 2001:41d0:fe89:5f00::1.2384 > dns2.isp.ovh.net.53: [udp sum ok] 5990+ A? addons-pa.clients6.google.com. (47)
16:43:42.617543 IP6 (flowlabel 0x9c17d, hlim 64, next-header UDP (17) payload length: 55) 2001:41d0:fe89:5f00::1.2384 > dns2.isp.ovh.net.53: [udp sum ok] 5990+ A? addons-pa.clients6.google.com. (47)
16:43:42.642609 IP6 (flowlabel 0x91f19, hlim 64, next-header UDP (17) payload length: 55) 2001:41d0:fe89:5f00::1.2384 > dns1.isp.ovh.net.53: [udp sum ok] 5990+ A? addons-pa.clients6.google.com. (47)
16:43:42.642637 IP6 (flowlabel 0x53a2d, hlim 64, next-header UDP (17) payload length: 55) 2001:41d0:fe89:5f00::1.2384 > dns2.isp.ovh.net.53: [udp sum ok] 5990+ A? addons-pa.clients6.google.com. (47)
16:43:42.642655 IP6 (flowlabel 0x9c17d, hlim 64, next-header UDP (17) payload length: 55) 2001:41d0:fe89:5f00::1.2384 > dns2.isp.ovh.net.53: [udp sum ok] 5990+ A? addons-pa.clients6.google.com. (47)
16:43:42.749122 IP (tos 0x0, ttl 64, id 4114, offset 0, flags [DF], proto UDP (17), length 117)
    10.166.179.2.54027 > 173.ip-167-114-241.eu.22837: [udp sum ok] UDP, length 89
16:43:43.813285 IP6 (flowlabel 0x6ae68, hlim 64, next-header ICMPv6 (58) payload length: 88) 2001:41d0:fe89:5f00::1 > par21s03-in-x0a.1e100.net: [icmp6 sum ok] ICMP6, destination unreachable, unreachable address 2001:41d0:fe89:5f04:8d0a:5eca:62fd:ea5e
[... cut ...]
16:43:57.478928 IP6 (flowlabel 0x43b39, hlim 64, next-header UDP (17) payload length: 45) 2001:41d0:fe89:5f00::1.49640 > dns2.isp.ovh.net.53: [udp sum ok] 54023+ A? eu.patch.battle.net. (37)
16:43:57.478951 IP6 (flowlabel 0x82609, hlim 64, next-header UDP (17) payload length: 45) 2001:41d0:fe89:5f00::1.49640 > dns2.isp.ovh.net.53: [udp sum ok] 54023+ A? eu.patch.battle.net. (37)
16:43:57.694029 IP (tos 0x0, ttl 64, id 4908, offset 0, flags [DF], proto UDP (17), length 117)
    10.166.179.2.54027 > 173.ip-167-114-241.eu.22837: [udp sum ok] UDP, length 89
64 packets captured
66 packets received by filter
2 packets dropped by kernel
```

angt, 2020-05-22:
Timeout.

closed 2020-05-22 by angt